YDLIDAR/ydlidar_ros2_driver #2: Services

k3street, 2020-09-11:
I am not able to utilize the start and stop scan services, when I list ros2 service list:
/ydlidar_ros2_driver_node/change_state
/ydlidar_ros2_driver_node/describe_parameters
/ydlidar_ros2_driver_node/get_parameter_types
/ydlidar_ros2_driver_node/get_parameters
/ydlidar_ros2_driver_node/list_parameters
/ydlidar_ros2_driver_node/set_parameters
/ydlidar_ros2_driver_node/set_parameters_atomically

I don't see start and stop scan

yangfuyuan, 2021-01-31:
The driver package has been updated,
1. Add stop_scan and start_scan services

Services list:
```
/launch_ros_13626/describe_parameters
/launch_ros_13626/get_parameter_types
/launch_ros_13626/get_parameters
/launch_ros_13626/list_parameters
/launch_ros_13626/set_parameters
/launch_ros_13626/set_parameters_atomically
/start_scan
/static_tf_pub_laser/describe_parameters
/static_tf_pub_laser/get_parameter_types
/static_tf_pub_laser/get_parameters
/static_tf_pub_laser/list_parameters
/static_tf_pub_laser/set_parameters
/static_tf_pub_laser/set_parameters_atomically
/stop_scan
/ydlidar_ros2_driver_node/change_state
/ydlidar_ros2_driver_node/describe_parameters
/ydlidar_ros2_driver_node/get_parameter_types
/ydlidar_ros2_driver_node/get_parameters
/ydlidar_ros2_driver_node/list_parameters
/ydlidar_ros2_driver_node/set_parameters
/ydlidar_ros2_driver_node/set_parameters_atomically
```

k3street, 2021-02-01:
Thank you for adding this service, my work round did teach me a few things but I am sure it is not the way I would have gone long-term. Thanks again!