YDLIDAR/ydlidar_ros2_driver #17: Launch files not working on Ubuntu 20.04

mkerklaan, 2022-05-14:
Hey all,

I'm trying to use the YDLidar on Ubuntu20.04 using ROS2. However, I keep getting the following error. I tried fixing it myself but I can't figure it out. I feel like it has something to do with using Ubuntu20.04 which builds differently somehow. You guys have any suggestions?

```
ros2 launch ydlidar_ros2_driver ydlidar_launch_view.py 
[INFO] [launch]: All log files can be found below /home/mkerklaan/.ros/log/2022-05-14-10-52-04-025192-mkerklaan-ThinkPad-T15-Gen-2i-603152
[INFO] [launch]: Default logging verbosity is set to INFO
Task exception was never retrieved
future: <Task finished name='Task-2' coro=<LaunchService._process_one_event() done, defined at /opt/ros/galactic/lib/python3.8/site-packages/launch/launch_service.py:226> exception=InvalidLaunchFileError('py')>
Traceback (most recent call last):
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_description_sources/any_launch_file_utilities.py", line 53, in get_launch_description_from_any_launch_file
    return loader(launch_file_path)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_description_sources/python_launch_file_utilities.py", line 68, in get_launch_description_from_python_launch_file
    return getattr(launch_file_module, 'generate_launch_description')()
  File "/home/mkerklaan/Documents/GitRepos/YLidar/install/ydlidar_ros2_driver/share/ydlidar_ros2_driver/launch/ydlidar_launch_view.py", line 40, in generate_launch_description
    driver_node = LifecycleNode(
TypeError: __init__() missing 2 required keyword-only arguments: 'name' and 'namespace'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_service.py", line 228, in _process_one_event
    await self.__process_event(next_event)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_service.py", line 248, in __process_event
    visit_all_entities_and_collect_futures(entity, self.__context))
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
    sub_entities = entity.visit(context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/action.py", line 108, in visit
    return self.execute(context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/actions/include_launch_description.py", line 130, in execute
    launch_description = self.__launch_description_source.get_launch_description(context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_description_source.py", line 84, in get_launch_description
    self._get_launch_description(self.__expanded_location)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_description_sources/any_launch_description_source.py", line 53, in _get_launch_description
    return get_launch_description_from_any_launch_file(location)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_description_sources/any_launch_file_utilities.py", line 56, in get_launch_description_from_any_launch_file
    raise InvalidLaunchFileError(extension, likely_errors=exceptions)
launch.invalid_launch_file_error.InvalidLaunchFileError: Caught exception when trying to load file of format [py]: __init__() missing 2 required keyword-only arguments: 'name' and 'namespace'
```

vik748, 2024-01-15:
+1 I am seeing the same error on ROS IRON, did you ever solve it?

vik748, 2024-01-16:
For those it might help, switching to the Humble branch resolved all the issues.

X-lanni, 2025-03-15:
In ROS2 Galactic and later versions:

`LifecycleNode` parameters have changed. You can no longer use `node_executable`, `node_name`, `node_namespace`.

Instead, you must use `executable`, `name`, and `namespace`.

After I modified these parameters, I was able to run the lidar launch.py ​​file normally.

```python
#!/usr/bin/python3
# Copyright 2020, EAIBOT

from ament_index_python.packages import get_package_share_directory

from launch import LaunchDescription
from launch_ros.actions import LifecycleNode
from launch_ros.actions import Node
from launch.actions import DeclareLaunchArgument
from launch.substitutions import LaunchConfiguration

import os

def generate_launch_description():
    share_dir = get_package_share_directory('ydlidar_ros2_driver')
    parameter_file = LaunchConfiguration('params_file')
    node_name = 'ydlidar_ros2_driver_node'

    params_declare = DeclareLaunchArgument(
        'params_file',
        default_value=os.path.join(share_dir, 'params', 'X2.yaml'),
        description='Path to the ROS2 parameters file to use.'
    )

    driver_node = LifecycleNode(
        package='ydlidar_ros2_driver',
        executable='ydlidar_ros2_driver_node',   # Updated: node_executable -> executable
        name='ydlidar_node',                     # Updated: node_name -> name
        namespace='/',                           # Updated: node_namespace -> namespace
        output='screen',
        emulate_tty=True,
        parameters=[parameter_file]
    )

    tf2_node = Node(
        package='tf2_ros',
        executable='static_transform_publisher',  # Updated: node_executable -> executable
        name='static_tf_pub_laser',               # Updated: node_name -> name
        arguments=['0.065', '0.0', '0.08', '3.14159265', '0.0', '0.0', '/base_footprint', '/laser_frame']
    )

    return LaunchDescription([
        params_declare,
        driver_node,
        tf2_node,
    ])
```