YDLIDAR/ydlidar_ros2_driver #21: Build Error in ROS2 Humble: no matching function -  rclcpp::Node::declare_parameter(const char [5])

slowrunner, 2022-10-22:
Compile errors in ROS2 Humble (on Ubuntu 22.04 Server Jammy)

```
$ colcon build --symlink-install
Starting >>> ydlidar_ros2_driver
[Processing: ydlidar_ros2_driver]                             
--- stderr: ydlidar_ros2_driver                               
/home/ubuntu/ydlidar_ros2_ws/src/ydlidar_ros2_driver/src/ydlidar_ros2_driver_node.cpp: In function ‘int main(int, char**)’:
/home/ubuntu/ydlidar_ros2_ws/src/ydlidar_ros2_driver/src/ydlidar_ros2_driver_node.cpp:44:26: error: no matching function for call to ‘rclcpp::Node::declare_parameter(const char [5])’
   44 |   node->declare_parameter("port");
      |   ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~
...

ubuntu@ROS2HH:~/ydlidar_ros2_ws$ printenv | grep ROS
ROS_VERSION=2
ROS_PYTHON_VERSION=3
ROS_DOMAIN_ID=0
ROS_LOCALHOST_ONLY=0
ROS_DISTRO=humble

ubuntu@ROS2HH:~/ydlidar_ros2_ws$ uname -a
Linux ROS2HH 5.15.0-1017-raspi #19-Ubuntu SMP PREEMPT Fri Oct 14 08:22:47 UTC 2022 aarch64 aarch64 aarch64 GNU/Linux

```

slowrunner, 2022-10-22:
Workaround:  Added  ", optvalue", ", f_optvalue", ", b_optvalue" as appropriate for each declare_parameter() to get it to properly compile.

Also had to change ydlidar_launch.py removing node_ from executable, name, namespace to get it to build:
```
 driver_node = LifecycleNode(package='ydlidar_ros2_driver',
                                executable='ydlidar_ros2_driver_node',
                                name='ydlidar_ros2_driver_node',
                                output='screen',
                                emulate_tty=True,
                                parameters=[parameter_file],
                                namespace='/',
                                )
```

MrBearing, 2022-10-25:
maybe, @lghrainbow is solve this issue ?

https://github.com/YDLIDAR/ydlidar_ros2_driver/pull/20

MrBearing, 2022-10-25:
In my enviroment, [this](https://github.com/lghrainbow/ydlidar_ros2_driver/tree/humble-devel) repository is work.

slowrunner, 2022-10-25:
In my environment, [this](https://github.com/lghrainbow/ydlidar_ros2_driver/tree/humble-devel) repository gives the following errors:
[ydlidar_ros2_driver_node-1] terminate called after throwing an instance of 'rclcpp::exceptions::UninitializedStaticallyTypedParameterException'
[ydlidar_ros2_driver_node-1]   what():  Statically typed parameter 'fixed_resolution' must be initialized.
[static_transform_publisher-2] [INFO] [1666711686.062313321] [static_tf_pub_laser]: Spinning until stopped - publishing transform

I had to change "resolution_fixed: true" to "fixed_resolution: true" in params/ydlidar.yaml and then the @lghrainbow humble updates worked for me

MrBearing, 2022-10-29:
Oh, sorry for my misunderstanding. 
You use raspberry pi ? 
It is different from my environment.(amd64 , Ubuntu 22.04)

rekabuk, 2022-11-21:
Mr Bearing #20 worked for me 😀

JHPHELAN, 2023-02-19:
@slowrunner "Workaround: Added ", optvalue", ", f_optvalue", ", b_optvalue" as appropriate for each declare_parameter() to get it to properly compile."
I'm a ROS2 newbie and don't know what 'optvalue' is nor which is appropriate where.  Could you please give example code changes?  Thanks!!

slowrunner, 2023-03-17:
This is my version that compiles under ROS2 Humble / Ubuntu 22.04 Jammy / Python 3.10.6

https://github.com/slowrunner/ROS2-GoPiGo3/blob/main/ros2ws/src/ydlidar_ros2_driver/src/ydlidar_ros2_driver_node.cpp.mine

closed 2023-03-17 by slowrunner

JHPHELAN, 2023-04-16:
@slowrunner 
Appreciate the update!!  
Copied/pasted your ydlidar_ros2_driver_node.cpp and recompiled with "unused parameter" warnings but didn't crash:
```
ubuntu@AUDACITY:~$ colcon build --symlink-install
Starting >>> ydlidar_ros2_driver
Starting >>> ydlidar_sdk
Finished <<< ydlidar_sdk [1.58s]
--- stderr: ydlidar_ros2_driver
/home/ubuntu/ydlidar_ros2_driver/src/ydlidar_ros2_driver_node.cpp: In lambda function:
/home/ubuntu/ydlidar_ros2_driver/src/ydlidar_ros2_driver_node.cpp:165:54: warning: unused parameter ‘request_header’ [-Wunused-parameter]
  165 |     [&laser](const std::shared_ptr<rmw_request_id_t> request_header,
      |              ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~
/home/ubuntu/ydlidar_ros2_driver/src/ydlidar_ros2_driver_node.cpp:166:56: warning: unused parameter ‘req’ [-Wunused-parameter]
  166 |   const std::shared_ptr<std_srvs::srv::Empty::Request> req,
      |   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~
/home/ubuntu/ydlidar_ros2_driver/src/ydlidar_ros2_driver_node.cpp:167:51: warning: unused parameter ‘response’ [-Wunused-parameter]
  167 |   std::shared_ptr<std_srvs::srv::Empty::Response> response) -> bool
      |   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~
/home/ubuntu/ydlidar_ros2_driver/src/ydlidar_ros2_driver_node.cpp: In lambda function:
/home/ubuntu/ydlidar_ros2_driver/src/ydlidar_ros2_driver_node.cpp:175:54: warning: unused parameter ‘request_header’ [-Wunused-parameter]
  175 |     [&laser](const std::shared_ptr<rmw_request_id_t> request_header,
      |              ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~
/home/ubuntu/ydlidar_ros2_driver/src/ydlidar_ros2_driver_node.cpp:176:56: warning: unused parameter ‘req’ [-Wunused-parameter]
  176 |   const std::shared_ptr<std_srvs::srv::Empty::Request> req,
      |   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~
/home/ubuntu/ydlidar_ros2_driver/src/ydlidar_ros2_driver_node.cpp:177:51: warning: unused parameter ‘response’ [-Wunused-parameter]
  177 |   std::shared_ptr<std_srvs::srv::Empty::Response> response) -> bool
      |   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~
---
Finished <<< ydlidar_ros2_driver [24.7s]

Summary: 2 packages finished [26.1s]
  1 package had stderr output: ydlidar_ros2_driver
```

JHP

ly-bo, 2024-04-15:
I'm using the same environment and getting the same error.
I later made a change to the code, and the compilation passed without a hitch.
` std::string str_optvalue = "/dev/ydlidar";
  std::string str_port = "port";
  node->declare_parameter(str_port,str_optvalue);
  node->get_parameter("port", str_optvalue);`