YDLIDAR/ydlidar_ros2_driver #22: lots of builds errors RO2 Humble in raspberry PI4

FrGe2016, 2023-01-04:
Many build errors when i build the package
I am not strong at cpp so here is are typical exemples ( there is many so much of them that i can not copy them all)

  391 |   declare_parameter(
      |   ^~~~~~~~~~~~~~~~~
/opt/ros/humble/include/rclcpp/rclcpp/node.hpp:391:3: note:   candidate expects 4 arguments, 1 provided
/home/mower/ydlidar_ws/src/ydlidar_ros2_ws/src/ydlidar_ros2_driver/src/ydlidar_ros2_driver_node.cpp:56:26: error: no matching function for call to ‘rclcpp::Node::declare_parameter(const char [9])’
   56 |   node->declare_parameter("frame_id");
      |   ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~
In file included from /opt/ros/humble/include/rclcpp/rclcpp/executors/single_threaded_executor.hpp:28,
                 from /opt/ros/humble/include/rclcpp/rclcpp/executors.hpp:22,
                 from /opt/ros/humble/include/rclcpp/rclcpp/rclcpp.hpp:155,
                 from /home/mower/ydlidar_ws/src/ydlidar_ros2_ws/src/ydlidar_ros2_driver/src/ydlidar_ros2_driver_node.cpp:23:
/opt/ros/humble/include/rclcpp/rclcpp/node.hpp:421:3: note: candidate: ‘template<class ParameterT> auto rclcpp::Node::declare_parameter(const string&, const ParameterT&, const ParameterDescriptor&, bool)’
  421 |   declare_parameter(
      |   ^~~~~~~~~~~~~~~~~
/opt/ros/humble/include/rclcpp/rclcpp/node.hpp:421:3: note:   template argument deduction/substitution failed:
/home/mower/ydlidar_ws/src/ydlidar_ros2_ws/src/ydlidar_ros2_driver/src/ydlidar_ros2_driver_node.cpp:56:26: note:   candidate expects 4 arguments, 1 provided
   56 |   node->declare_parameter("frame_id");
      |   ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~
In file included from /opt/ros/humble/include/rclcpp/rclcpp/executors/single_threaded_executor.hpp:28,
                 from /opt/ros/humble/include/rclcpp/rclcpp/executors.hpp:22,
                 from /opt/ros/humble/include/rclcpp/rclcpp/rclcpp.hpp:155,
                 from /home/mower/ydlidar_ws/src/ydlidar_ros2_ws/src/ydlidar_ros2_driver/src/ydlidar_ros2_driver_node.cpp:23:
/opt/ros/humble/include/rclcpp/rclcpp/node.hpp:434:3: note: candidate: ‘template<class ParameterT> auto rclcpp::Node::declare_parameter(const string&, const ParameterDescriptor&, bool)’

cclngit, 2023-01-06:
Hello, have you been able to find a solution to the problem you mentioned on Humble? I have also encountered the same issue on multiple devices including `Khadas vim4`, `Ubuntu 22.04 VM`, `RP4` and `RP3`, but have been unable to resolve it.

FrGe2016, 2023-01-07:
no Not solved yet. If i have development i will let you know. Please to the
same if you have somen positive results

cclngit, 2023-01-07:
Hello @FrGe2016 , I have already debugged and rewritten some code which you can see in pull request #23 (this is just a draft). It worked on my Khadas vim4, but I have not tested it on my Raspberry Pi 3/4 or my Ubuntu virtual machine.

FrGe2016, 2023-01-07:
Thanks, I will check it out.

FrGe2016, 2023-01-15:
Thanks it is compiling But i have a new issue. The lidar is stopping rapidly after the first tests. 
        I have adjusted the yaml file for the X4 model
        I have pretty much the same result testing only the SDK.
        Using the same rpi4 with ubuntu 20 and ROS1 is working well?

[ydlidar_ros2_driver_node-1] LiDAR successfully connected
[ydlidar_ros2_driver_node-1] [YDLIDAR]:Lidar running correctly ! The health status: good
[ydlidar_ros2_driver_node-1] LiDAR init success, Elapsed time 633 ms
[ydlidar_ros2_driver_node-1] [YDlidar] Create thread 0x936488E0
[ydlidar_ros2_driver_node-1] [CYdLidar] Successed to start scan mode, Elapsed time 1065 ms
[ydlidar_ros2_driver_node-1] timeout count: 1
[ydlidar_ros2_driver_node-1] timeout count: 2
[ydlidar_ros2_driver_node-1] timeout count: 1
[ydlidar_ros2_driver_node-1] 0x936488E0 thread has been canceled
[ydlidar_ros2_driver_node-1] [CYdLidar] Failed to turn on the Lidar, because the lidar is [Device Failed].
[ydlidar_ros2_driver_node-1] [INFO] [1673791571.975928676] [ydlidar_ros2_driver_node]: [YDLIDAR INFO] Now YDLIDAR is stopping .......
[INFO] [ydlidar_ros2_driver_node-1]: process has finished cleanly [pid 5357]

cclngit, 2023-01-15:
@FrGe2016 I don't know why but I figured out that the yaml parameters did not work, you must change the parameters directly in the source code, I explain it in the [README](https://github.com/cclngit/ydlidar_ros2_driver#configure-lidar), then change the parameters in the source code and re-use the colcon build command in the same directory like you did before a step forward to recompile the code.

cclngit, 2023-01-15:
The SDK did not work for me too

FrGe2016, 2023-01-15:
I have made a few additional tests

From the ydlidar site i have downlodad the SDK for the X4. Now the sdk is alone is working. But with the ros 2 driver the scan is stopping after around 10 seconds. I will look at your read me to modify the parameters and let you know

cclngit, 2023-01-16:
The sdk never worked for me, but after some troubleshooting changing directly  `node->get_parameter("/dev/ttyUSB0", str_optvalue);` with the right parameters work for me. If what I guess is correct when we type `ros2 launch ydlidar_ros2_driver ydlidar_launch.py` we may add the parameters after like `-p port dev/ttyUSB0 -p bauderate 115200` for example but with my version that no longer work anymore.

camelator, 2023-02-28:
Same issue on rockpi

camelator, 2023-03-01:
workaround here https://github.com/YDLIDAR/ydlidar_ros2_driver/issues/21 worked for me! ...