YDLIDAR/ydlidar_ros2_driver #25: How to slam using YDLidar

hyng809, 2023-01-20:
Hello
This is the first time using YDLidar.
I have verified that Lidar works through Rviz on Ubuntu.
I want to create maps and use navigation using Lidar, which I am currently using.
I would like to know more detailed process and tutorial about this.
This is the development environment I am currently developing.
- Ubuntu 20.04
- ROS2 foxy
-YDLidar TG15
- Raspberry Pi 4
-PC

anasderkaoui, 2023-11-30:
If you have succeeded at creating maps with this config of yours please let me know. I am currently working on the same subject but in need of more tutorials explaining how to generate a map using the ydlidar (mine is G4) on ros2 foxy.

ndp-seee-hust, 2024-06-03:
Hello @anasderkaoui. Thank for sharing!
I have followed your instructions and I have a problem. My map cannot update or scale when I rotate and move the Lidar. Please help me fix this problem!
[image]

anasderkaoui, 2024-06-03:
Hello @ndp-seee-hust, make sure that you have the odometry node running.
This was, in my case, one of the main reasons the map did not update. And also, move the lidar very slowly and don't rotate it suddenly because that will overlap the maps that will be created.
I advise you to take a look at [my repo](https://github.com/anasderkaoui/AutoRCX/tree/9a5f9122718a348c6f09c48475e196802984838f/SLAM) for more detailed information.

NiVick103, 2025-11-28:
> Try other links and github related.
> I have just seen [here](https://github.com/nodehubs/ydlidar_ros_driver) that the default one might be 230400, another common one might be 115200

thanks，I will try again.

musstt, 2026-04-09:
> * Make sure you install "slam_toolbox" (package to do SLAM): `sudo apt update && sudo apt install ros-foxy-slam-toolbox`
> 
>     * Turn on the LiDAR: `ros2 launch ydlidar_ros2_driver ydlidar_launch_view.py` (This also launches the tf tree required for slam, but you have to change "base_link" in [this line](https://github.com/YDLIDAR/ydlidar_ros2_driver/blob/c2acbfa32ed5465def97a0190208e96bcba5ce94/launch/ydlidar_launch_view.py#L50) to "base_footprint" and rebuild the package)
> 
>     * In another terminal, run: `ros2 run tf2_ros static_transform_publisher 0 0 0 0 0 0 odom base_footprint` **Do this only if you don't have an odometry output from like wheel encoders...**
> 
>     * Now run in another terminal: `ros2 launch slam_toolbox online_sync_launch.py` This is just an example, there are other options, you can view them by double pressing the tab key. After that, you should see "Registering Sensor..."
> 
>     * Finally, run RVIZ2 and add the topic "map", it takes a moment to load the map but it eventually shows up!!

Hello,

I would like to ask about SLAM behavior when using only a LiDAR without any additional sensors.

Do we need odometry data that is actually changing (dynamic odom) in order for the map to update properly?

Here is my setup:

* Ubuntu 22.04
* ROS 2 Humble
* YDLIDAR X4
* Only LiDAR (no encoders, no IMU, no additional sensors)

I have verified that the LiDAR is working correctly in RViz and the /scan topic is publishing properly.

I tried running:

ros2 run tf2_ros static_transform_publisher 0 0 0 0 0 0 odom base_footprint

With this, the map seems to update once at the beginning. However, when I physically move or rotate the LiDAR, the map does not update, does not expand, and does not even overlap at all. It just stays exactly the same as the initial state.

So my question is:
Is it required to have a changing (dynamic) odometry input for SLAM (especially with slam_toolbox) to properly update and expand the map?

I followed the setup exactly as described, but without real odometry, the map does not behave as expected.

Thank you!