YDLIDAR/ydlidar_ros2_driver #28: Error, cannot retrieve YDLidar health code: ffffffff

jtoobias, 2023-09-07:
Hi, so I have just started implementing a ROS 2 project using a ydlidar x2l LiDAR.
When I run the ydlidar_launch_view.py file i get the following output:
I noticed a similar issue that was closed, but making adjustments didn't help me (unless i did it wrong).

ros2 launch ydlidar_ros2_driver ydlidar_launch_view.py
[INFO] [launch]: Default logging verbosity is set to INFO
<launch.substitutions.launch_configuration.LaunchConfiguration object at 0x7f589dfb3250>
[INFO] [ydlidar_ros2_driver_node-1]: process started with pid [5837]
[INFO] [static_transform_publisher-2]: process started with pid [5839]
[static_transform_publisher-2] [WARN] [1694086841.296649159] []: Old-style arguments are deprecated; see --help for new-style arguments
[ydlidar_ros2_driver_node-1] [INFO] [1694086841.318627688] [ydlidar_node]: [YDLIDAR INFO] Current ROS Driver Version: 1.0.1
[ydlidar_ros2_driver_node-1] 
[ydlidar_ros2_driver_node-1] [YDLIDAR] SDK initializing
[ydlidar_ros2_driver_node-1] [YDLIDAR] SDK has been initialized
[ydlidar_ros2_driver_node-1] [YDLIDAR] SDK Version: 1.1.18
[static_transform_publisher-2] [INFO] [1694086841.326665310] [static_tf_pub_laser]: Spinning until stopped - publishing transform
[static_transform_publisher-2] translation: ('0.000000', '0.000000', '0.020000')
[static_transform_publisher-2] rotation: ('0.000000', '0.000000', '0.000000', '1.000000')
[static_transform_publisher-2] from 'base_link' to 'laser_frame'
[ydlidar_ros2_driver_node-1] [YDLIDAR] Lidar successfully connected [/dev/serial/by-id/usb-Silicon_Labs_CP2102_USB_to_UART_Bridge_Controller_0001-if00-port0:115200]
[ydlidar_ros2_driver_node-1] [YDLIDAR] Error, cannot retrieve YDLidar health code: ffffffff
[ydlidar_ros2_driver_node-1] [YDLIDAR] Fail to get baseplate device information
[ydlidar_ros2_driver_node-1] [YDLIDAR] Lidar init success, Elapsed time 2662 ms
[ydlidar_ros2_driver_node-1] [YDLIDAR] Failed to start scan mode: ffffffff
[ydlidar_ros2_driver_node-1] [INFO] [1694086849.624063705] [ydlidar_node]: [YDLIDAR INFO] Now YDLIDAR is stopping .......
[INFO] [ydlidar_ros2_driver_node-1]: process has finished cleanly [pid 5837]

When I launch the file, the LiDAR rotation speed drops down aswell. I have check the voltage during this time and its not sagging, I have a supply with more than enough power for the LiDAR.

My launch file:
  `
def generate_launch_description():
    share_dir = get_package_share_directory('ydlidar_ros2_driver')
    parameter_file = LaunchConfiguration('params_file')
    node_name = 'ydlidar_ros2_driver_node'

    params_declare = DeclareLaunchArgument('params_file',
                                            default_value=os.path.join(
                                            share_dir, 'params', 'X2.yaml'),
                                            description='FPath to the ROS2 parameters file to use.')
    
    print (parameter_file)
    
    driver_node = LifecycleNode(package='ydlidar_ros2_driver',
                                executable='ydlidar_ros2_driver_node',
                                name='ydlidar_node',
                                output='screen',
                                emulate_tty=True,
                                parameters=[parameter_file],
                                namespace='/',
                                )
    tf2_node = Node(package='tf2_ros',
                    executable='static_transform_publisher',
                    name='static_tf_pub_laser',
                    arguments=['0', '0', '0.02','0', '0', '0', '1','base_link','laser_frame'],
                    )

    return LaunchDescription([
        params_declare,
        driver_node,
        tf2_node,
    ])`

Parameter file used:
`ydlidar_ros2_driver_node:
  ros__parameters:
    port: /dev/ydlidar
    frame_id: laser_frame
    ignore_array: ""
    baudrate: 115200
    lidar_type: 1
    device_type: 0
    sample_rate: 3
    abnormal_check_count: 4
    fixed_resolution: true
    reversion: false
    inverted: false
    auto_reconnect: true
    isSingleChannel: true
    intensity: false
    support_motor_dtr: true
    angle_max: 180.0
    angle_min: -180.0
    range_max: 12.0
    range_min: 0.1
    frequency: 10.0
    invalid_range_is_inf: false`

Any help is greatly appreciated. Thank you in advance

jtoobias, 2023-09-07:
Ok I fixed this, I think it was just a serial port issue (wrong port).
I put my launch and parameter file here anyway to help others on ROS Humble: I am using the ydlidar 2xl
`#!/usr/bin/python3
# Copyright 2020, EAIBOT
# Licensed under the Apache License, Version 2.0 (the "License");
# you may not use this file except in compliance with the License.
# You may obtain a copy of the License at
#
#     http://www.apache.org/licenses/LICENSE-2.0
#
# Unless required by applicable law or agreed to in writing, software
# distributed under the License is distributed on an "AS IS" BASIS,
# WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
# See the License for the specific language governing permissions and
# limitations under the License.

from ament_index_python.packages import get_package_share_directory

from launch import LaunchDescription
from launch_ros.actions import LifecycleNode
from launch_ros.actions import Node
from launch.actions import DeclareLaunchArgument
from launch.substitutions import LaunchConfiguration
from launch.actions import LogInfo

import lifecycle_msgs.msg
import os

def generate_launch_description():
    share_dir = get_package_share_directory('ydlidar_ros2_driver')
    rviz_config_file = os.path.join(share_dir, 'config','ydlidar.rviz')
    parameter_file = LaunchConfiguration('params_file')
    # node_name = 'ydlidar_ros2_driver_node'

    params_declare = DeclareLaunchArgument('params_file',
                                           default_value=os.path.join(
                                               share_dir, 'params', 'X2.yaml'),
                                           description='FPath to the ROS2 parameters file to use.')

    driver_node = LifecycleNode(package='ydlidar_ros2_driver',
                                executable='ydlidar_ros2_driver_node',
                                name='ydlidar_ros2_driver_node',
                                output='screen',
                                emulate_tty=True,
                                parameters=[parameter_file],
                                namespace='/',
                                )
    tf2_node = Node(package='tf2_ros',
                    executable='static_transform_publisher',
                    name='static_tf_pub_laser',
                    arguments=['0', '0', '0.02','0', '0', '0', '1','base_link','laser_frame'],
                    )
    
    rviz2_node = Node(package='rviz2',
                    executable='rviz2',
                    name='rviz2',
                    arguments=['-d', rviz_config_file],
                    )

    return LaunchDescription([
        params_declare,
        driver_node,
        tf2_node,
        rviz2_node
    ])
`

Parameter file:
'ydlidar_ros2_driver_node:
  ros__parameters:
    port: /dev/serial/by-id/usb-Silicon_Labs_CP2102_USB_to_UART_Bridge_Controller_0001-if00-port0
    frame_id: laser_frame
    ignore_array: ""
    baudrate: 115200
    lidar_type: 1
    device_type: 0
    sample_rate: 3
    abnormal_check_count: 4
    fixed_resolution: true
    reversion: false
    inverted: false
    auto_reconnect: true
    isSingleChannel: true
    intensity: false
    support_motor_dtr: true
    angle_max: 180.0
    angle_min: -180.0
    range_max: 12.0
    range_min: 0.1
    frequency: 10.0
    invalid_range_is_inf: false
'

closed 2023-09-07 by jtoobias

fadai333, 2024-09-20:
Thanks, I had the same problem, but my solution was to change the driver_node in driver_node from "yalidar" to "/".

fadai333, 2024-09-20:
Thanks, I had the same problem, but my solution was to change the driver_node in driver_node from 'yalidar' to '/', which could be one reason for the problem