YDLIDAR/ydlidar_ros2_driver #29: [ERROR] [ydlidar_ros2_driver_node-1]: process has died, exit code -6

hfcaio, 2023-09-07:
when i run `ros2 launch ydlidar_ros2_driver ydlidar_launch.py` and `ros2 launch ydlidar_ros2_driver ydlidar_launch_view.py` both of then give me this error

[image]

Ahtisham123Ahmed, 2024-03-19:
@hfcaio did you find out solution?

jungwoonghyun, 2024-07-03:
@hfcaio Hi did you find out solution?

jungwoonghyun, 2024-07-03:
@Ahtisham123Ahmed Did you find out solution?

Ahtisham123Ahmed, 2024-07-03:
> @Ahtisham123Ahmed Did you find out solution?

I didnt remember the issue exactly but i figure out the problem and use ydlidarx3pro for my robot

jungwoonghyun, 2024-07-05:
@Ahtisham123Ahmed Anyway there is some way to fix it. Thanks to let me know

joIcecube, 2024-11-23:
Hi everyone. any update ? thx all

Ahtisham123Ahmed, 2024-11-27:
> Hi everyone. any update ? thx all

@joIcecube Yes, I used the YDLidar X3 Pro for my robot. I think I fixed the issue. I can share the file for launching YDLidar and some important settings. There were some issues in the file that needed to be updated.