YDLIDAR/ydlidar_ros2_driver #33: Not working with Humble or later

flochre, 2023-10-28:
Hi guys,

I was experiencing problem with the branch working on Humble, and I am not too fan of the solution in the humble branch because all the files are executable with is useless and not a good practice.

So I created a few pull request to change this.
Please see here the pull requests opened today

#30 (this one reused #19)
#31 
#32 

If nobody wants to merge them you can use my repo.. all pull request are merged in my main branch : https://github.com/flochre/ydlidar_ros2_driver

flochre, 2023-10-28:
I added a PR #34 to get Rviz Launch working also

umicar1234, 2024-03-29:
i am still not able to build this in my RPI 4 works on ubuntu 22.0 ros2 iron

sanjana1999, 2024-05-17:
> i am still not able to build this in my RPI 4 works on ubuntu 22.0 ros2 iron

Facing the same issue. Did you find any solution?

umicar1234, 2024-05-18:
Yes, I solved it, it only support upto ros2 humble, only few changes have to be done in c++ file