YDLIDAR/ydlidar_ros2_driver #39: Build error in ROS2 rolling version with the master branch

ywiyogo, 2023-12-25:
I run in my dockerfile this command
```
RUN mkdir -p /home/${USERNAME}/ros2_ws/src && cd /home/${USERNAME}/ros2_ws/src \
  && git clone https://github.com/YDLIDAR/ydlidar_ros2_driver.git -b master\
  && cd .. && . /opt/ros/rolling/setup.bash \
  && rosdep update && rosdep install -y -r --from-paths src --ignore-src \
  && colcon build --symlink-install
````

and got this compiler error:
````
17.78 /home/ros/ros2_ws/src/ydlidar_ros2_driver/src/ydlidar_ros2_driver_node.cpp: In function ‘int main(int, char**)’:
17.78 /home/ros/ros2_ws/src/ydlidar_ros2_driver/src/ydlidar_ros2_driver_node.cpp:44:26: error: no matching function for call to ‘rclcpp::Node::declare_parameter(const char [5])’
17.78    44 |   node->declare_parameter("port");
17.78       |   ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~
````

Is the master branch compatible to the ROS2 rolling?

I can build the humble branch without any issue.

jieunko, 2023-12-28:
There is a branch humble
Maybe you can try that branch too

ywiyogo, 2023-12-29:
@jieunko yes, as I wrote already, I can build the branch humble. But, for the master branch, which ROS version is it suitable for?