YDLIDAR/ydlidar_ros2_driver #42: Wrong parameter type, parameter {invalid_range_is_inf} is of type {double}, setting it to {bool} is not allowed.

TheMasterProgrammer, 2024-02-02:
hello i have x3 lidar (which is similar to x2) and I am using the ydlidar sdk on ubuntu 22 with ros2 humble implementation. Inside the ydlidar.yaml,  whenever i set invalid_range_is_inf to false i get the error:
parameter 'invalid_range_is_inf' has invalid type: Wrong parameter type, parameter {invalid_range_is_inf} is of type {double}, setting it to {bool} is not allowed.

which indicates that the type is double.

But when i change it to 0.0 i get the error:
what(): parameter 'invalid_range_is_inf' has invalid type: expected [bool] got [double]

this is a paradox and i need help
thanks

shreyavsanoj7, 2024-06-28:
did u get a soln? Im kinda facing the same issue..
thanx

TheMasterProgrammer, 2024-06-28:
@shreyavsanoj7 I did. i had to modify the cpp source code of the ydlidar and i removed the part of the code that referenced invalid_range_is_inf

closed 2024-06-28 by TheMasterProgrammer