YDLIDAR/ydlidar_ros2_driver #44: Colcon build failed

amalshaji4540, 2024-02-24:
I am getting this error message

amal@amal-Aspire-A515-57G:~/ydlidar$ colcon build --symlink-install
Starting >>> ydlidar_ros2_driver
--- stderr: ydlidar_ros2_driver                             
/home/amal/ydlidar/src/ydlidar_ros2_driver/src/ydlidar_ros2_driver_node.cpp: In function ‘int main(int, char**)’:
/home/amal/ydlidar/src/ydlidar_ros2_driver/src/ydlidar_ros2_driver_node.cpp:44:26: error: no matching function for call to ‘rclcpp::Node::declare_parameter(const char [5])’
   44 |   node->declare_parameter("port");
      |   ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~
In file included from /opt/ros/humble/include/rclcpp/rclcpp/executors/single_threaded_executor.hpp:28,
                 from /opt/ros/humble/include/rclcpp/rclcpp/executors.hpp:22,
                 from /opt/ros/humble/include/rclcpp/rclcpp/rclcpp.hpp:155,
                 from /home/amal/ydlidar/src/ydlidar_ros2_driver/src/ydlidar_ros2_driver_node.cpp:23:
/opt/ros/humble/include/rclcpp/rclcpp/node.hpp:421:3: note: candidate: ‘template<class ParameterT> auto rclcpp::Node::declare_parameter(const string&, const ParameterT&, const ParameterDescriptor&, bool)’
  421 |   declare_parameter(
      |   ^~~~~~~~~~~~~~~~~
/opt/ros/humble/include/rclcpp/rclcpp/node.hpp:421:3: note:   template argument deduction/substitution failed:
/home/amal/ydlidar/src/ydlidar_ros2_driver/src/ydlidar_ros2_driver_node.cpp:44:26: note:   candidate expects 4 arguments, 1 provided
   44 |   node->declare_parameter("port");
      |   ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~
In file included from /opt/ros/humble/include/rclcpp/rclcpp/executors/single_threaded_executor.hpp:28,
                 from /opt/ros/humble/include/rclcpp/rclcpp/executors.hpp:22,
                 from /opt/ros/humble/include/rclcpp/rclcpp/rclcpp.hpp:155,
                 from /home/amal/ydlidar/src/ydlidar_ros2_driver/src/ydlidar_ros2_driver_node.cpp:23:
/opt/ros/humble/include/rclcpp/rclcpp/node.hpp:434:3: note: candidate: ‘template<class ParameterT> auto rclcpp::Node::declare_parameter(const string&, const ParameterDescriptor&, bool)’
  434 |   declare_parameter(
      |   ^~~~~~~~~~~~~~~~~
/opt/ros/humble/include/rclcpp/rclcpp/node.hpp:434:3: note:   template argument deduction/substitution failed:
/home/amal/ydlidar/src/ydlidar_ros2_driver/src/ydlidar_ros2_driver_node.cpp:44:26: note:   couldn’t deduce template parameter ‘ParameterT’
   44 |   node->declare_parameter("port");
      |   ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~
In file included from /opt/ros/humble/include/rclcpp/rclcpp/executors/single_threaded_executor.hpp:28,
                 from /opt/ros/humble/include/rclcpp/rclcpp/executors.hpp:22,
                 from /opt/ros/humble/include/rclcpp/rclcpp/rclcpp.hpp:155,
                 from /home/amal/ydlidar/src/ydlidar_ros2_driver/src/ydlidar_ros2_driver_node.cpp:23:
/opt/ros/humble/include/rclcpp/rclcpp/node.hpp:366:3: note: candidate: ‘const rclcpp::ParameterValue& rclcpp::Node::declare_parameter(const string&, const rclcpp::ParameterValue&, const ParameterDescriptor&, bool)’
  366 |   declare_parameter(
      |   ^~~~~~~~~~~~~~~~~
/opt/ros/humble/include/rclcpp/rclcpp/node.hpp:366:3: note:   candidate expects 4 arguments, 1 provided
/opt/ros/humble/include/rclcpp/rclcpp/node.hpp:391:3: note: candidate: ‘const rclcpp::ParameterValue& rclcpp::Node::declare_parameter(const string&, rclcpp::ParameterType, const ParameterDescriptor&, bool)’
  391 |   declare_parameter(
      |   ^~~~~~~~~~~~~~~~~
/opt/ros/humble/include/rclcpp/rclcpp/node.hpp:391:3: note:   candidate expects 4 arguments, 1 provided
/home/amal/ydlidar/src/ydlidar_ros2_driver/src/ydlidar_ros2_driver_node.cpp:51:26: error: no matching function for call to ‘rclcpp::Node::declare_parameter(const char [13])’
   51 |   node->declare_parameter("ignore_array");
      |   ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~
In file included from /opt/ros/humble/include/rclcpp/rclcpp/executors/single_threaded_executor.hpp:28,
                 from /opt/ros/humble/include/rclcpp/rclcpp/executors.hpp:22,
                 from /opt/ros/humble/include/rclcpp/rclcpp/rclcpp.hpp:155,
                 from /home/amal/ydlidar/src/ydlidar_ros2_driver/src/ydlidar_ros2_driver_node.cpp:23:
/opt/ros/humble/include/rclcpp/rclcpp/node.hpp:421:3: note: candidate: ‘template<class ParameterT> auto rclcpp::Node::declare_parameter(const string&, const ParameterT&, const ParameterDescriptor&, bool)’
  421 |   declare_parameter(
      |   ^~~~~~~~~~~~~~~~~
/opt/ros/humble/include/rclcpp/rclcpp/node.hpp:421:3: note:   template argument deduction/substitution failed:
/home/amal/ydlidar/src/ydlidar_ros2_driver/src/ydlidar_ros2_driver_node.cpp:51:26: note:   candidate expects 4 arguments, 1 provided
   51 |   node->declare_parameter("ignore_array");
      |   ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~
In file included from /opt/ros/humble/include/rclcpp/rclcpp/executors/single_threaded_executor.hpp:28,
                 from /opt/ros/humble/include/rclcpp/rclcpp/executors.hpp:22,
                 from /opt/ros/humble/include/rclcpp/rclcpp/rclcpp.hpp:155,
                 from /home/amal/ydlidar/src/ydlidar_ros2_driver/src/ydlidar_ros2_driver_node.cpp:23:
/opt/ros/humble/include/rclcpp/rclcpp/node.hpp:434:3: note: candidate: ‘template<class ParameterT> auto rclcpp::Node::declare_parameter(const string&, const ParameterDescriptor&, bool)’
  434 |   declare_parameter(
      |   ^~~~~~~~~~~~~~~~~
/opt/ros/humble/include/rclcpp/rclcpp/node.hpp:434:3: note:   template argument deduction/substitution failed:
/home/amal/ydlidar/src/ydlidar_ros2_driver/src/ydlidar_ros2_driver_node.cpp:51:26: note:   couldn’t deduce template parameter ‘ParameterT’
   51 |   node->declare_parameter("ignore_array");
      |   ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~
In file included from /opt/ros/humble/include/rclcpp/rclcpp/executors/single_threaded_executor.hpp:28,
                 from /opt/ros/humble/include/rclcpp/rclcpp/executors.hpp:22,
[... 629 lines cut ...]
                 from /opt/ros/humble/include/rclcpp/rclcpp/rclcpp.hpp:155,
                 from /home/amal/ydlidar/src/ydlidar_ros2_driver/src/ydlidar_ros2_driver_node.cpp:23:
/opt/ros/humble/include/rclcpp/rclcpp/node.hpp:366:3: note: candidate: ‘const rclcpp::ParameterValue& rclcpp::Node::declare_parameter(const string&, const rclcpp::ParameterValue&, const ParameterDescriptor&, bool)’
  366 |   declare_parameter(
      |   ^~~~~~~~~~~~~~~~~
/opt/ros/humble/include/rclcpp/rclcpp/node.hpp:366:3: note:   candidate expects 4 arguments, 1 provided
/opt/ros/humble/include/rclcpp/rclcpp/node.hpp:391:3: note: candidate: ‘const rclcpp::ParameterValue& rclcpp::Node::declare_parameter(const string&, rclcpp::ParameterType, const ParameterDescriptor&, bool)’
  391 |   declare_parameter(
      |   ^~~~~~~~~~~~~~~~~
/opt/ros/humble/include/rclcpp/rclcpp/node.hpp:391:3: note:   candidate expects 4 arguments, 1 provided
/home/amal/ydlidar/src/ydlidar_ros2_driver/src/ydlidar_ros2_driver_node.cpp:144:26: error: no matching function for call to ‘rclcpp::Node::declare_parameter(const char [10])’
  144 |   node->declare_parameter("range_min");
      |   ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~
In file included from /opt/ros/humble/include/rclcpp/rclcpp/executors/single_threaded_executor.hpp:28,
                 from /opt/ros/humble/include/rclcpp/rclcpp/executors.hpp:22,
                 from /opt/ros/humble/include/rclcpp/rclcpp/rclcpp.hpp:155,
                 from /home/amal/ydlidar/src/ydlidar_ros2_driver/src/ydlidar_ros2_driver_node.cpp:23:
/opt/ros/humble/include/rclcpp/rclcpp/node.hpp:421:3: note: candidate: ‘template<class ParameterT> auto rclcpp::Node::declare_parameter(const string&, const ParameterT&, const ParameterDescriptor&, bool)’
  421 |   declare_parameter(
      |   ^~~~~~~~~~~~~~~~~
/opt/ros/humble/include/rclcpp/rclcpp/node.hpp:421:3: note:   template argument deduction/substitution failed:
/home/amal/ydlidar/src/ydlidar_ros2_driver/src/ydlidar_ros2_driver_node.cpp:144:26: note:   candidate expects 4 arguments, 1 provided
  144 |   node->declare_parameter("range_min");
      |   ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~
In file included from /opt/ros/humble/include/rclcpp/rclcpp/executors/single_threaded_executor.hpp:28,

amalshaji4540, 2024-02-24:
I'm using ros2 humble
The error is solved by using humble branch of ydlidar_ros2_driver.
I used this command
```git clone -b humble https://github.com/YDLIDAR/ydlidar_ros2_driver.git```

UdaySMukherjee, 2024-06-18:
Yeah I'm getting the same issue while building Tortoisebot

arambarricalvoj, 2025-01-30:
@amalshaji4540 your solution works! However I get the following warnings. Is it ok?

`Starting >>> ydlidar_ros2_driver
--- stderr: ydlidar_ros2_driver                                
/home/javierac/ydlidar_ros2_ws/src/ydlidar_ros2_driver/src/ydlidar_ros2_driver_node.cpp: In lambda function:
/home/javierac/ydlidar_ros2_ws/src/ydlidar_ros2_driver/src/ydlidar_ros2_driver_node.cpp:169:54: warning: unused parameter ‘request_header’ [-Wunused-parameter]
  169 |     [&laser](const std::shared_ptr<rmw_request_id_t> request_header,
      |              ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~
/home/javierac/ydlidar_ros2_ws/src/ydlidar_ros2_driver/src/ydlidar_ros2_driver_node.cpp:170:56: warning: unused parameter ‘req’ [-Wunused-parameter]
  170 |   const std::shared_ptr<std_srvs::srv::Empty::Request> req,
      |   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~
/home/javierac/ydlidar_ros2_ws/src/ydlidar_ros2_driver/src/ydlidar_ros2_driver_node.cpp:171:51: warning: unused parameter ‘response’ [-Wunused-parameter]
  171 |   std::shared_ptr<std_srvs::srv::Empty::Response> response) -> bool
      |   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~
/home/javierac/ydlidar_ros2_ws/src/ydlidar_ros2_driver/src/ydlidar_ros2_driver_node.cpp: In lambda function:
/home/javierac/ydlidar_ros2_ws/src/ydlidar_ros2_driver/src/ydlidar_ros2_driver_node.cpp:179:54: warning: unused parameter ‘request_header’ [-Wunused-parameter]
  179 |     [&laser](const std::shared_ptr<rmw_request_id_t> request_header,
      |              ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~
/home/javierac/ydlidar_ros2_ws/src/ydlidar_ros2_driver/src/ydlidar_ros2_driver_node.cpp:180:56: warning: unused parameter ‘req’ [-Wunused-parameter]
  180 |   const std::shared_ptr<std_srvs::srv::Empty::Request> req,
      |   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~
/home/javierac/ydlidar_ros2_ws/src/ydlidar_ros2_driver/src/ydlidar_ros2_driver_node.cpp:181:51: warning: unused parameter ‘response’ [-Wunused-parameter]
  181 |   std::shared_ptr<std_srvs::srv::Empty::Response> response) -> bool
      |   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~
---
Finished <<< ydlidar_ros2_driver [13.9s]

Summary: 1 package finished [14.2s]
  1 package had stderr output: ydlidar_ros2_driver
`

James6588, 2025-03-01:
@arambarricalvoj I got the same warning. Did you find out if the warnings causes any problem? Please share. Thanks.

arambarricalvoj, 2025-03-01:
I don't remember because I switched to [https://github.com/Oyefusi-Samuel/ydlidar_ros2_driver-master](https://github.com/Oyefusi-Samuel/ydlidar_ros2_driver-master), which is adapted to ROS2 Humble and works perfectly. Check issue: [#1] (https://github.com/Oyefusi-Samuel/ydlidar_ros2_driver-master/issues/1)