YDLIDAR/ydlidar_ros2_driver #46: Getting an error while running 'ydlidar_launch.py'

ugur-cakici, 2024-07-19:
Hello,

I am trying to run the ydlidar_ros2_driver package using the ydlidar_launch.py file but I am encountering some errors. Below is the error message I receive:
Error Message for ydlidar_launch.py:

ugur@ugur-pc:~/lidar_deneme_ws$ ros2 launch ydlidar_ros2_driver ydlidar_launch.py 
[INFO] [launch]: All log files can be found below /home/ugur/.ros/log/2024-07-19-15-53-11-904326-ugur-pc-36555
[INFO] [launch]: Default logging verbosity is set to INFO
2024-07-19 15:53:12.006 [RTPS_TRANSPORT_SHM Error] Failed init_port fastrtps_port7411: open_and_lock_file failed -> Function open_port_internal
[INFO] [ydlidar_ros2_driver_node-1]: process started with pid [36567]
[INFO] [static_transform_publisher-2]: process started with pid [36569]
[static_transform_publisher-2] [WARN] [1721393592.045122172] []: Old-style arguments are deprecated; see --help for new-style arguments
[ydlidar_ros2_driver_node-1] 2024-07-19 15:53:12.048 [RTPS_TRANSPORT_SHM Error] Failed init_port fastrtps_port7411: open_and_lock_file failed -> Function open_port_internal
[ydlidar_ros2_driver_node-1] [INFO] [1721393592.060761613] [ydlidar_ros2_driver_node]: [YDLIDAR INFO] Current ROS Driver Version: 1.0.1
[ydlidar_ros2_driver_node-1] 
[ydlidar_ros2_driver_node-1] [YDLIDAR] SDK initializing
[ydlidar_ros2_driver_node-1] [YDLIDAR] SDK has been initialized
[ydlidar_ros2_driver_node-1] [YDLIDAR] SDK Version: 1.2.6
[static_transform_publisher-2] [INFO] [1721393592.065746091] [static_tf_pub_laser]: Spinning until stopped - publishing transform
[static_transform_publisher-2] translation: ('0.000000', '0.000000', '0.020000')
[static_transform_publisher-2] rotation: ('0.000000', '0.000000', '0.000000', '1.000000')
[static_transform_publisher-2] from 'base_link' to 'laser_frame'
[ydlidar_ros2_driver_node-1] [YDLIDAR] Lidar successfully connected [/dev/ttyUSB0:230400]
[ydlidar_ros2_driver_node-1] [YDLIDAR] Error, cannot retrieve YDLidar health code: ffffffff
[ydlidar_ros2_driver_node-1] [YDLIDAR] Fail to get baseplate device information
[ydlidar_ros2_driver_node-1] [YDLIDAR] Lidar init success, Elapsed time 2693 ms
[ydlidar_ros2_driver_node-1] [YDLIDAR] Failed to start scan mode: ffffffff
[ydlidar_ros2_driver_node-1] [INFO] [1721393600.391924264] [ydlidar_ros2_driver_node]: [YDLIDAR INFO] Now YDLIDAR is stopping .......
[INFO] [ydlidar_ros2_driver_node-1]: process has finished cleanly [pid 36567]

Additionally, I also tried running the ydlidar_launch_view.py file and encountered a similar issue. Here is the error message for that:

Error Message for ydlidar_launch_view.py:

ugur@ugur-pc:~/lidar_deneme_ws$ ros2 launch ydlidar_ros2_driver ydlidar_launch_view.py 
[INFO] [launch]: All log files can be found below /home/ugur/.ros/log/2024-07-19-15-53-18-910451-ugur-pc-36591
[INFO] [launch]: Default logging verbosity is set to INFO
2024-07-19 15:53:19.038 [RTPS_TRANSPORT_SHM Error] Failed init_port fastrtps_port7411: open_and_lock_file failed -> Function open_port_internal
[INFO] [ydlidar_ros2_driver_node-1]: process started with pid [36603]
[INFO] [static_transform_publisher-2]: process started with pid [36605]
[INFO] [rviz2-3]: process started with pid [36607]
[static_transform_publisher-2] [WARN] [1721393599.084757134] []: Old-style arguments are deprecated; see --help for new-style arguments
[ydlidar_ros2_driver_node-1] 2024-07-19 15:53:19.089 [RTPS_TRANSPORT_SHM Error] Failed init_port fastrtps_port7411: open_and_lock_file failed -> Function open_port_internal
[rviz2-3] Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
[static_transform_publisher-2] [INFO] [1721393599.107133858] [static_tf_pub_laser]: Spinning until stopped - publishing transform
[static_transform_publisher-2] translation: ('0.000000', '0.000000', '0.020000')
[static_transform_publisher-2] rotation: ('0.000000', '0.000000', '0.000000', '1.000000')
[static_transform_publisher-2] from 'base_link' to 'laser_frame'
[ydlidar_ros2_driver_node-1] [INFO] [1721393599.108111083] [ydlidar_ros2_driver_node]: [YDLIDAR INFO] Current ROS Driver Version: 1.0.1
[ydlidar_ros2_driver_node-1] 
[ydlidar_ros2_driver_node-1] [YDLIDAR] SDK initializing
[ydlidar_ros2_driver_node-1] [YDLIDAR] SDK has been initialized
[ydlidar_ros2_driver_node-1] [YDLIDAR] SDK Version: 1.2.6
[ydlidar_ros2_driver_node-1] [YDLIDAR] Lidar successfully connected [/dev/ttyUSB0:230400]
[rviz2-3] [INFO] [1721393599.484946166] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-3] [INFO] [1721393599.485190091] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-3] [INFO] [1721393599.506348921] [rviz2]: Stereo is NOT SUPPORTED
[ydlidar_ros2_driver_node-1] [YDLIDAR] Error, cannot retrieve YDLidar health code: ffffffff
[ydlidar_ros2_driver_node-1] [YDLIDAR] Fail to get baseplate device information
[ydlidar_ros2_driver_node-1] [YDLIDAR] Lidar init success, Elapsed time 2671 ms
[ydlidar_ros2_driver_node-1] [YDLIDAR] Failed to start scan mode: ffffffff
[ydlidar_ros2_driver_node-1] [INFO] [1721393607.414753895] [ydlidar_ros2_driver_node]: [YDLIDAR INFO] Now YDLIDAR is stopping .......
[INFO] [ydlidar_ros2_driver_node-1]: process has finished cleanly [pid 36603]

Could you please help me understand what might be causing these issues and how to resolve them?

Thank you.

joIcecube, 2024-11-23:
Hi everyone. any update ? thx all

ashemah, 2025-01-07:
You need to pass the params_file param pointing to the correct file for your lidar. 

e.g. ros2 launch ydlidar_ros2_driver ydlidar_launch.py params_file:=<path to params yaml>

The params file for each lidar type is in the params dir of the package.