YDLIDAR/ydlidar_ros2_driver #51: compile error

Oyefusi-Samuel, 2025-01-24:
434 |   declare_parameter(
      |   ^~~~~~~~~~~~~~~~~
/opt/ros/humble/include/rclcpp/rclcpp/node.hpp:434:3: note:   template argument deduction/substitution failed:
/home/sam/ydlidar_ros2_ws/src/ydlidar_ros2_driver/src/ydlidar_ros2_driver_node.cpp:154:26: note:   couldn’t deduce template parameter ‘ParameterT’
  154 |   node->declare_parameter("invalid_range_is_inf");
      |   ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~
In file included from /opt/ros/humble/include/rclcpp/rclcpp/executors/single_threaded_executor.hpp:28,
                 from /opt/ros/humble/include/rclcpp/rclcpp/executors.hpp:22,
                 from /opt/ros/humble/include/rclcpp/rclcpp/rclcpp.hpp:155,
                 from /home/sam/ydlidar_ros2_ws/src/ydlidar_ros2_driver/src/ydlidar_ros2_driver_node.cpp:23:
/opt/ros/humble/include/rclcpp/rclcpp/node.hpp:366:3: note: candidate: ‘const rclcpp::ParameterValue& rclcpp::Node::declare_parameter(const string&, const rclcpp::ParameterValue&, const ParameterDescriptor&, bool)’
  366 |   declare_parameter(
      |   ^~~~~~~~~~~~~~~~~
/opt/ros/humble/include/rclcpp/rclcpp/node.hpp:366:3: note:   candidate expects 4 arguments, 1 provided
/opt/ros/humble/include/rclcpp/rclcpp/node.hpp:391:3: note: candidate: ‘const rclcpp::ParameterValue& rclcpp::Node::declare_parameter(const string&, rclcpp::ParameterType, const ParameterDescriptor&, bool)’
  391 |   declare_parameter(
      |   ^~~~~~~~~~~~~~~~~
/opt/ros/humble/include/rclcpp/rclcpp/node.hpp:391:3: note:   candidate expects 4 arguments, 1 provided
/home/sam/ydlidar_ros2_ws/src/ydlidar_ros2_driver/src/ydlidar_ros2_driver_node.cpp: In lambda function:
/home/sam/ydlidar_ros2_ws/src/ydlidar_ros2_driver/src/ydlidar_ros2_driver_node.cpp:168:54: warning: unused parameter ‘request_header’ [-Wunused-parameter]
  168 |     [&laser](const std::shared_ptr<rmw_request_id_t> request_header,
      |              ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~
/home/sam/ydlidar_ros2_ws/src/ydlidar_ros2_driver/src/ydlidar_ros2_driver_node.cpp:169:56: warning: unused parameter ‘req’ [-Wunused-parameter]
  169 |   const std::shared_ptr<std_srvs::srv::Empty::Request> req,
      |   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~
/home/sam/ydlidar_ros2_ws/src/ydlidar_ros2_driver/src/ydlidar_ros2_driver_node.cpp:170:51: warning: unused parameter ‘response’ [-Wunused-parameter]
  170 |   std::shared_ptr<std_srvs::srv::Empty::Response> response) -> bool
      |   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~
/home/sam/ydlidar_ros2_ws/src/ydlidar_ros2_driver/src/ydlidar_ros2_driver_node.cpp: In lambda function:
/home/sam/ydlidar_ros2_ws/src/ydlidar_ros2_driver/src/ydlidar_ros2_driver_node.cpp:178:54: warning: unused parameter ‘request_header’ [-Wunused-parameter]
  178 |     [&laser](const std::shared_ptr<rmw_request_id_t> request_header,
      |              ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~
/home/sam/ydlidar_ros2_ws/src/ydlidar_ros2_driver/src/ydlidar_ros2_driver_node.cpp:179:56: warning: unused parameter ‘req’ [-Wunused-parameter]
  179 |   const std::shared_ptr<std_srvs::srv::Empty::Request> req,
      |   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~
/home/sam/ydlidar_ros2_ws/src/ydlidar_ros2_driver/src/ydlidar_ros2_driver_node.cpp:180:51: warning: unused parameter ‘response’ [-Wunused-parameter]
  180 |   std::shared_ptr<std_srvs::srv::Empty::Response> response) -> bool
      |   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~
gmake[2]: *** [CMakeFiles/ydlidar_ros2_driver_node.dir/build.make:76: CMakeFiles/ydlidar_ros2_driver_node.dir/src/ydlidar_ros2_driver_node.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:139: CMakeFiles/ydlidar_ros2_driver_node.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< ydlidar_ros2_driver [1.77s, exited with code 2]

Summary: 0 packages finished [1.84s]
  1 package failed: ydlidar_ros2_driver
  1 package had stderr output: ydlidar_ros2_driver
sam@sam:~/ydlidar_ros2_ws$

arambarricalvoj, 2025-01-30:
I have the same problem and found this link: https://qiita.com/Yuya-Shimizu/items/c516b076ecc15864c0c5.
However, I haven't tested it yet, as it takes some hours of hard work.

Oyefusi-Samuel, 2025-02-01:
@arambarricalvoj I solved the problem myself, it was challenging but rewarding. I wrote the codes for both the driver and client node. You can check here https://github.com/Oyefusi-Samuel/ydlidar_ros2_driver-master.git

Oyefusi-Samuel, 2025-02-01:
> I have the same problem and found this link: https://qiita.com/Yuya-Shimizu/items/c516b076ecc15864c0c5. However, I haven't tested it yet, as it takes some hours of hard work.

@arambarricalvoj i tried that also dint work, there are some deprecated functions in there not compatible with humble