YDLIDAR/ydlidar_ros2_driver #53: Error, cannot retrieve Lidar health code -1

arambarricalvoj, 2025-02-05:
After having trouble to compile the package I run the launcher and get this error. Points cloud doesn't appear in RVIZ2. Anyone can help me?

[image]

arambarricalvoj, 2025-02-06:
solved using [https://github.com/Oyefusi-Samuel/ydlidar_ros2_driver-master](https://github.com/Oyefusi-Samuel/ydlidar_ros2_driver-master) repo from Oyefusi-Samuel.

If working with ydlidar x4 also see [https://github.com/Oyefusi-Samuel/ydlidar_ros2_driver-master/issues/1](https://github.com/Oyefusi-Samuel/ydlidar_ros2_driver-master/issues/1)

tolongshiz, 2025-03-31:
do you solved it?i have the same problem

arambarricalvoj, 2025-06-02:
Maybe it's quite late, but I wanted to let you know it's a parameter issue. Check your lidar's parameters and modify them in the code or params file. For instance, in [https://github.com/Oyefusi-Samuel/ydlidar_ros2_driver-master/issues/1](https://github.com/Oyefusi-Samuel/ydlidar_ros2_driver-master/issues/1) I tell what changes should be done for YDLIDAR X4 model.