YDLIDAR/ydlidar_ros2_driver #56: LiDAR is not visualized in rviz

mira-lily, 2025-04-03:
Hello everyone!
Please help me to fix the following. 
1. YDLidar SDK installed correctly
2. ws also was by built succesfully
3. Lidar works with data visualization application on Windows

OS: Ubuntu 24.04.1 LTS 
ROS2: jazzy

Thank you

[image]
[image]

tuclen3, 2025-04-03:
The problem may come from you run two command. You just run rviz

[image]

mira-lily, 2025-04-03:
> The problem may come from you run two command. You just run rviz
> 
> [image]

What do you mean? I only tried second way.
ros2 launch ydlidar_ros2_driver ydlidar_launch_view.py

mrRobot62, 2025-04-05:
Hi Mira-Lily, I saw a checksum error in your screenshot. Please refer to my issue under Ydlidar-SDK, you posted, the same issue.  (https://github.com/YDLIDAR/YDLidar-SDK/issues/63)

If you get this checksum error, your lidar ist not running and your will not see anything in rviz.

mira-lily, 2025-04-30:
This issue solved. download again the driver and install. Then check the 'params; is configured as a correct.

closed 2025-04-30 by mira-lily